xpack-dev-tools/arm-none-eabi-gcc-xpack #1: libgcov.a is no-op

Novakov, 2019-09-20:
`libgcov.a` included in 8.3.1 release contains only no-op functions.

Assembly dump of `__gcov_init` function from `lib\gcc\arm-none-eabi\8.3.1\thumb\v7-m\nofp\libgcov.a`:
```
00000000 <__gcov_init>:
   0:	4770      	bx	lr
   2:	bf00      	nop
```

Dump it the same for both Windows and Linux distribution package.

Official release of arm-none-eabi-gcc 8.3.1 from ARM contains full functions.

ilg-ul, 2019-09-20:
I'll investigate, I did not use libgcov before and I have to study a bit to see what went wrong during the build.

ilg-ul, 2019-10-07:
The issues is caused by a configuration error. In the second pass, when the system libraries are built, the `configure` tool checks if the system headers are available, otherwise it concludes it is a special build and defines `-Dinhibit_libc` which generally has the effect of creating empty library functions.

The actual check for the system headers is simple, only the presence of`stdio.h` is verified in the location specified by the `--with-sysroot` option. Usually this points to a full system, and by default the file is expected to be in the `/usr/include` folder. However, in our distribution the headers are in the `/include` folder, and the test fails.

The solution is to also add `--with-native-system-header-dir=/include` to define the correct location. 

A new release will be available soon.

ilg-ul, 2019-10-11:
Fixed in v8.3.1-1.2, on 2019-10-11.

ilg-ul, 2019-10-11:
@Novakov, please check the new release and let me know it the library also functional, since I do not have a test case for it.

Novakov, 2019-10-14:
@ilg-ul Just check release 8.3.1-1.2 and code coverage is working nicely. Thanks for the fix!

ilg-ul, 2019-10-14:
You're welcome!

ilg-ul, 2019-10-14:
Since checking code coverage for embedded systems is not a well known topic, could you contribute a post to the project forum documenting how you did it, so that others can benefit from your experience?

closed 2019-10-14 by ilg-ul

Novakov, 2019-10-16:
Most on this subject is covered here: https://mcuoneclipse.com/2014/12/26/code-coverage-for-embedded-target-with-eclipse-gcc-and-gcov/. Exact implemention depends on how someone is runing unt tests, we are using QEmu with semihosting so it's a lot easier than on hardware.

ilg-ul, 2019-10-16:
you mean you are still using those stubs and manually issuing debugger commands to write the files? they are totally useless, if you create semihosted projects with the f4discovery template, for instance, everything is handled automatically, you don‘t have to do anything but add the compiler and linker options.

Novakov, 2019-10-16:
No, I'm just calling `__gcov_flush` as the last instruction in `main()`. I linked that article because that's what I used when I was getting gcov running the first time and it describes how gcov works.

ilg-ul, 2019-10-16:
Try to simply return from `main()` with an error code as usual, flushing is done automatically at exit.

ilg-ul, 2019-10-16:
Do you have a program named `gcov` in your path?

Novakov, 2019-10-16:
In general case it might work, for me - it doesn't (`exit` in newlib is messing with exit code which is then returned by qemu and I rely on on that exit code).

If you ask if `arm-none-eabi-gcov` is in toolchain, then yes, it is.
I limit number of items in PATH, so even gcc is not there.

ilg-ul, 2019-10-16:
If you are using the semihosting code from the templates and not the librdimon, exit works as expected.

No, I ask about gcov. As far as I know, Eclipse does not know about arm-none-eabi-gcov, so if you can see the nice Eclipse results, I wonder how they are generated.

Novakov, 2019-10-16:
It's been a while since I last used Eclipse IDE, for viewing coverage report I'm using gcovr (it generates HTML reports)

ilg-ul, 2019-10-16:
Another good reason why it would be useful to have your experience in a forum article.