xpack-dev-tools/arm-none-eabi-gcc-xpack #2: doc folder is redundant

HarjitSi, 2019-10-13:
### Description

xPack\ARM Embedded GCC\8.3.1-1.2\doc

has substantially the same content as 

xPack\ARM Embedded GCC\8.3.1-1.2\share\doc

It seems xPack\ARM Embedded GCC\8.3.1-1.2\share\doc is a super set of xPack\ARM Embedded GCC\8.3.1-1.2\doc

### Steps to Reproduce

1. Do a diff between the folders

**Expected behaviour:** No duplication

**Actual behaviour:** Files are duplicated

### Versions

This is the file I found it in: https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/download/v8.3.1-1.2/xpack-arm-none-eabi-gcc-8.3.1-1.2-win32-x64.zip

ilg-ul, 2019-10-13:
I confirm the issue, probably to a bug in the Windows build script, there is redundant `doc` folder.

The macOS and GNU/Linux binaries seem ok.

ilg-ul, 2019-10-14:
The issue was fixed in the common helper scripts and will apply to all builds.

For consistency, the `doc` folder is located in the `share` folder on all platforms.

The fix will apply starting with the next release.

ilg-ul, 2019-11-02:
Fixed in v8.3.1-1.3, on 2019-11-02.

closed 2019-11-02 by ilg-ul