xpack-dev-tools/arm-none-eabi-gcc-xpack #3: On macOS, arm-none-eabi-gdb-py fails to initialise Python

ilg-ul, 2019-10-16:
```console
$ /Users/ilg/Library/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/8.3.1-1.2.1/.content/bin/arm-none-eabi-gdb-py --version
Traceback (most recent call last):
  File "/Users/ilg/Library/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/8.3.1-1.2.1/.content/arm-none-eabi/share/gdb/python/gdb/__init__.py", line 143, in auto_load_packages
    __import__(modname)
  File "/Users/ilg/Library/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/8.3.1-1.2.1/.content/arm-none-eabi/share/gdb/python/gdb/command/frame_filters.py", line 24, in <module>
    import gdb.frames
  File "/Users/ilg/Library/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/8.3.1-1.2.1/.content/arm-none-eabi/share/gdb/python/gdb/frames.py", line 23, in <module>
    import collections
  File "/Users/ilg/opt/xbb/lib/python2.7/collections.py", line 21, in <module>
    from operator import itemgetter as _itemgetter, eq as _eq
ImportError: dlopen(/Users/ilg/opt/xbb/lib/python2.7/lib-dynload/operator.so, 2): Symbol not found: _PyObject_Not
  Referenced from: /Users/ilg/opt/xbb/lib/python2.7/lib-dynload/operator.so
  Expected in: flat namespace
 in /Users/ilg/opt/xbb/lib/python2.7/lib-dynload/operator.so

GNU gdb (xPack GNU ARM Embedded GCC, 64-bit) 8.3.0.20190703-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

The Python 3 version seems ok:

```console
$ /Users/ilg/Library/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/8.3.1-1.2.1/.content/bin/arm-none-eabi-gdb-py3 --version
GNU gdb (xPack GNU ARM Embedded GCC, 64-bit) 8.3.0.20190703-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

The `PYTHONHOME` variable must point to a valid Python environment, where `lib/python2.7` or `lib/python3.7` must be present.

To use the system Python 2, the location is `/System/Library/Frameworks/Python.framework/Versions/2.7`; there is no system Python 3 in current macOS (10.14).

HarjitSi, 2019-10-16:
My use case is a little different than what you intended - I'm using your files but with Qt Creator. So, I take the xpack files and place them where I want.

In any case, I tried the Windows gdb with Python yesterday and it spewed issues with Python not finding a module when I tried Python 3.8 and 2.7.16. This morning I tried 2.7.13 and no issues.

This is just in case it is a similar issue.

ilg-ul, 2019-10-16:
> This morning I tried 2.7.13 and no issues

oops! I checked the scripts and indeed the Windows gdb-py was built with 2.7.13; it looks a compatibility issue with 2.7.16.

I'll try to build it again with 2.7.16 and see what happens.

Could you enter a separate issue so we can better track this?

ilg-ul, 2019-10-17:
Building and embedding Python 2 into an application (like GDB) seems a bit of black magic and I could not find a way to make it work with the 2.7.13 built in the mac XBB..

However, by switching the toolchain build to the Apple existing Python 2 (an old 2.7.10) seems to be a reasonable workaround. 

To be noted that for Python 3, which is not provided by Apple and is taken from the mac XBB, the `arm-none-eabi-gdb-py3` is fine.

ilg-ul, 2019-11-02:
Fixed in v8.3.1-1.3, on 2019-11-02.

closed 2019-11-02 by ilg-ul