xpack-dev-tools/arm-none-eabi-gcc-xpack #4: arm-none-eabi-gdb-py generates Python warnings with Python 2.7.16

HarjitSi, 2019-10-16:
### Description

I tried the Windows gdb with Python 3.8 and then Python 2.7.16 and it spewed issues with Python not finding a module. This morning I tried Python 2.7.13 and no issues.

### Steps to Reproduce

1. Install arm-none-eabi-gcc-xPack - v8.3.1-1.2
2. Install the "failing" versions of Python
3. Launch arm-none-eabi-gdb-py.exe

**Expected behaviour:** 
See no Python warning messages.

**Actual behaviour:** 
Sorry I didn't save them and don't want to reinstall the ones with the warnings. If you can't repro., then, I'll certainly repro. and include info. 

I do recall this:
ImportError: cannot import name _remove_dead_weakref

### Versions
- [Arm GCC version]
arm-none-eabi-gcc-xPack (x64 version) - v8.3.1-1.2

- [operating system]
Windows 10 x64 1903

ilg-ul, 2019-10-16:
the Windows gdb-py was built with 2.7.13; it looks a compatibility issue with 2.7.16.

I build it again with 2.7.16 please try the following binary (save the one you have and override it)

https://www.dropbox.com/s/tnomd7gh7zy305x/arm-none-eabi-gdb-py.exe?dl=0

ilg-ul, 2019-10-16:
I tried on my Win 10 VM and I got this:

```console
C:\Users\ilg>C:\Users\ilg\Desktop\xPack\ARM Embedded GCC\8.3.1-1.2b\bin\arm-none-eabi-gdb-py.exe: warning: Couldn't determine a path for the index cache directory.
Traceback (most recent call last):
  File "c:\users\ilg\desktop\xpack\arm embedded gcc\8.3.1-1.2b\arm-none-eabi\share\gdb/python\gdb\__init__.py", line 143, in auto_load_packages
    __import__(modname)
  File "c:\users\ilg\desktop\xpack\arm embedded gcc\8.3.1-1.2b\arm-none-eabi\share\gdb/python\gdb\command\frame_filters.py", line 21, in <module>
    import copy
  File "C:\Python27\Lib\copy.py", line 52, in <module>
    import weakref
  File "C:\Python27\Lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

Traceback (most recent call last):
  File "c:\users\ilg\desktop\xpack\arm embedded gcc\8.3.1-1.2b\arm-none-eabi\share\gdb/python\gdb\__init__.py", line 143, in auto_load_packages
    __import__(modname)
  File "c:\users\ilg\desktop\xpack\arm embedded gcc\8.3.1-1.2b\arm-none-eabi\share\gdb/python\gdb\command\pretty_printers.py", line 19, in <module>
    import copy
  File "C:\Python27\Lib\copy.py", line 52, in <module>
    import weakref
  File "C:\Python27\Lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

Traceback (most recent call last):
  File "c:\users\ilg\desktop\xpack\arm embedded gcc\8.3.1-1.2b\arm-none-eabi\share\gdb/python\gdb\__init__.py", line 143, in auto_load_packages
    __import__(modname)
  File "c:\users\ilg\desktop\xpack\arm embedded gcc\8.3.1-1.2b\arm-none-eabi\share\gdb/python\gdb\command\type_printers.py", line 17, in <module>
    import copy
  File "C:\Python27\Lib\copy.py", line 52, in <module>
    import weakref
  File "C:\Python27\Lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

GNU gdb (xPack GNU Arm Embedded GCC, 64-bit) 8.3.0.20190703-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

With 2.7.13 the result is (almost) ok:

```console
C:\Users\ilg>"C:\Users\ilg\Desktop\xPack\ARM Embedded GCC\8.3.1-1.2b\bin\arm-none-eabi-gdb-py.exe" --version
C:\Users\ilg\Desktop\xPack\ARM Embedded GCC\8.3.1-1.2b\bin\arm-none-eabi-gdb-py.exe: warning: Couldn't determine a path for the index cache directory.
GNU gdb (xPack GNU Arm Embedded GCC, 64-bit) 8.3.0.20190703-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

The Python 2.7 binaries were downloaded from https://www.python.org/downloads/ and installed in the default location `C:\Python27`.

ilg-ul, 2019-10-16:
The new binary compiled with 2.7.16 did not make any difference.

Further investigations are needed.

ilg-ul, 2019-10-16:
I tried 2.7.16/15/14 and all failed. Google returns lots of references to `ImportError: cannot import name _remove_dead_weakref`, but apparently the reported problems were fixed since 2.7.14, which does not seem the case.

The current workaround is to use 2.7.13.

HarjitSi, 2019-10-17:
I'm able to reproduce the results you observed with 2.7.13. Thanks for working through this. This is such a pain.

C:\Program Files\gccARM\bin\arm-none-eabi-gdb-py.exe: warning: Couldn't determine a path for the ind
ex cache directory.
GNU gdb (xPack GNU ARM Embedded GCC, 64-bit) 8.3.0.20190703-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb)

ilg-ul, 2019-10-18:
It looks like the culprit is the `bin\python27.dll` file, which was copied during build from the 2.7.13 libraries, and later on prevents using other versions.

I removed it and tried the existing `arm-none-eabi-gdb-py.exe` with all Pythons 2.7.13-16; all went well.

FYI, for testing purposes I have installed all 4 versions in separate folders and I am using the environment variable to select the desired one.

```console
C:\Users\ilg>set PYTHONHOME=C:\Python27.16

C:\Users\ilg>C:\Users\ilg\AppData\Roaming\xPacks\@xpack-dev-tools\arm-none-eabi-gcc\8.3.1-1.2.1\.content\bin\arm-none-eabi-gdb-py.exe --version
C:\Users\ilg\AppData\Roaming\xPacks\@xpack-dev-tools\arm-none-eabi-gcc\8.3.1-1.2.1\.content\bin\arm-none-eabi-gdb-py.exe: warning: Couldn't determine a path for the index cache directory.
GNU gdb (xPack GNU ARM Embedded GCC, 64-bit) 8.3.0.20190703-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

```

I will update the build script to prevent the dll to be copied in the distribution and make another release.

ilg-ul, 2019-11-02:
Fixed in v8.3.1-1.3, on 2019-11-02.

closed 2019-11-02 by ilg-ul