xpack-dev-tools/arm-none-eabi-gcc-xpack #6: The GDB Text User Interface

fivdi, 2020-01-22:
This ticket was opened after discussing the topic at [The xPack Forums](https://www.tapatalk.com/groups/xpack/the-gdb-text-user-interface-t38.html)

The GDB Text User Interface, TUI in short, is a terminal interface which uses the curses library to show the source file, the assembly output, the program registers and GDB commands in separate text windows. The TUI is available only when GDB is configured with the --enable-tui configure option. For further information see [here](https://ftp.gnu.org/old-gnu/Manuals/gdb/html_chapter/gdb_19.html).

It looks like the GDB executable (arm-none-eabi-gdb) provided by arm-none-eabi-gcc-xpack is not configured with option --enable-tui as the TUI is not available. Running the command "layout src" to display the source file in a separate text window displays a message saying that "layout" is an undefined command.
```
(gdb) layout src
Undefined command: "layout".  Try "help".
```
Is it correct that arm-none-eabi-gdb is not configured with the --enable-tui configure option? If so, would it be possible to change arm-none-eabi-gcc-xpack so that GDB is configured with the --enable-tui configure option making the TUI available?

ilg-ul, 2020-01-22:
> Is it correct that arm-none-eabi-gdb is not configured with the --enable-tui configure option?

I guess so, I don't remember using this configure option.

> would it be possible to change arm-none-eabi-gcc-xpack so that GDB is configured with the --enable-tui configure option making the TUI available?

In principle yes, but in practice it depends on how difficult is to bring in all required dependencies. I guess ncurses & co. must be also processed by the build script.

I'll try to consider it for the next release.

ilg-ul, 2020-03-07:
I added ncurses to the project, enabled TUI, built everything and now, after starting gdb, the C-x C-a magic sequence brings a ncurses frame in the terminal. I don't know where to go from here, but I guess the mission was accomplished. 

For now only the macOS binaries were built and tested. I'll proceed with Linux and Windows builds, and let you know when ready.

What platform are you using?

fivdi, 2020-03-08:
That's great news. Thank you very much.

> What platform are you using?

Linux (Ububtu 18.04, 32-bit)

```
$ uname -a
Linux linuxbox 4.4.0-174-generic #204-Ubuntu SMP Wed Jan 29 06:41:58 UTC 2020 i686 i686 i686 GNU/Linux
```

ilg-ul, 2020-03-19:
Please test the new experimental binaries available from:

https://github.com/xpack-dev-tools/pre-releases/releases/tag/experimental

The compiler does not include the multilibs, so don't use it yet, test only gdb if the tui interface is functional.

fivdi, 2020-03-19:
Thank you for working on this.

I'm afraid I can't get it to work. This is what I see when an attempt is made to display the source file in a separate text window:
```
(gdb) layout src
Cannot enable the TUI: error opening terminal [TERM=xterm-256color]
```
Here's a complete debugging session
```
$ arm-none-eabi-gdb
GNU gdb (xPack GNU Arm Embedded GCC, 32-bit) 8.3.0.20190709-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=i686-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x08000254 in ?? ()
(gdb) file blinky.elf
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from blinky.elf...
(gdb) monitor reset
(gdb) monitor halt
target halted due to debug-request, current mode: Thread 
xPSR: 0x81000000 pc: 0x080001bc msp: 0x20004fe0
(gdb) load
Loading section .text, size 0x2e4 lma 0x8000000
Start address 0x8000254, load size 740
Transfer rate: 3 KB/sec, 740 bytes/write.
(gdb) b main
Breakpoint 1 at 0x800017a: file blinky.c, line 14.
(gdb) conti
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.

Breakpoint 1, main () at blinky.c:14
14	  gpio_setup();
(gdb) layout src
Cannot enable the TUI: error opening terminal [TERM=xterm-256color]
(gdb) 
```
What I've also noticed is that hitting the backspace key actually prints a space.

ilg-ul, 2020-03-19:
Most probably I overlooked some configuration options, `ncurses` is particularly tricky to build.

I'll review it and come back with a version that at least accepts this command. 

I remember that during tests, at a certain moment, this command worked, I'm 100% sure I saw a window in the terminal, so it should be possible to bring it back.

ilg-ul, 2020-03-24:
Brian,

It proved that adding `--tui` support to GDB is more tricky than expected. The problem is that TUI uses `libncurses`, which is highly dependent on the local `terminfo` configuration.

The first hard dependency is the folder where the `terminfo` data is located, which must be defined during `ncurses` configuration. As recommended, I used `/usr/share/terminfo`. If your system has a different location... I don't know, perhaps there is an environment variable to define it.

Then `ncurses` has several incompatible versions and lots of configuration options. I used the latest one, v6.2, and only a few options, which looked reasonable.

The result is that my binary distribution is able to start properly with `--tui` on Manjaro 19, but fails on Ubuntu 18.04 LTS with the same error that you reported.

In these cases one workaround is to remove the `libncurses.so.*` from my distribution, and install (possibly compile from sources) a ncurses 6 package on your system.

The configuration I used is:

https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/blob/dc8470e6d4768bdf464efbcf278ae28c12c2fc74/scripts/container-libs-functions-source.sh#L1049

I will do some more tests, perhaps I can identify some missed detail.

If you have any suggestions on how to move this forward, I would appreciate them.

The new binaries are available from the same experimental url.

ilg-ul, 2020-03-24:
The 64-bit binaries run without problems on openSUSE Tumbleweed.

ilg-ul, 2020-03-24:
After checking how the `ncurses` package is built on Ubuntu, I discovered that the `terminfo` folder settings is different.

It is possible to set the search path with `TERMINFO_DIRS`.

With the following setting, gdb started with `--tui` on Ubuntu 20:

```
export TERMINFO_DIRS=/etc/terminfo:/lib/terminfo:/usr/share/terminfo
```

I'll change the settings in my `ncurses` configuration.

ilg-ul, 2020-03-25:
I fixed the hardwired terminfo path to the above and now gdb switches to TUI mode on Ubuntu too.

Please test it on your machine and confirm.

fivdi, 2020-03-25:
Excellent job Liviu, it's working :smile: 

[image]

I didn't have to modify the search path `TERMINFO_DIRS`.

In a comment above I mentioned an issue where a backspace was printed as a space. This issue has also been resolved.

Once again, thank you very much for taking the time to work on this.

ilg-ul, 2020-03-25:
You're welcome!

fivdi, 2020-07-02:
The GDB Text User Interface works well for me with [xPack GNU Arm Embedded GCC v9.3.1-1.1](https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/tag/v9.3.1-1.1).

ilg-ul, 2020-07-03:
Great! Thank you for your feedback.

The new binaries are under testing now, the release will be public soon.

ilg-ul, 2020-07-03:
Fixed in v9.3.1-1.1.

closed 2020-07-03 by ilg-ul

EdwinFairchild, 2022-04-07:
any chance of getting this to work on windows?

ilg-ul, 2022-04-07:
I don't know if this is even possible on Windows. If you know of a distribution that does this, please let me know, and we can study the build scripts, perhaps the same solution can be applied here.