xpack-dev-tools/arm-none-eabi-gcc-xpack #7: C++ Exception Support isn't working (possibly because newlib-nano is incorrectly selected).

willson556, 2020-08-24:
### Description

It seems that in the xPack distribution of ARM GCC, C++ exception support isn't working. The symptoms are the same as when using `newlib-nano` instead of `newlib`.

### Steps to Reproduce

1. Install a recent `arm-none-eabi-gcc` xPack (this reproduced with all 4 of the most recent releases) as well as the `emu-arm` xPack.
2. Clone the repro project from [willson556/cortex-m-exception-test](https://github.com/willson556/cortex-m-exception-test) and open in Eclipse Embedded CDT.
3. Set the project _ARM Toolchains Paths_ and QEMU paths to their respective xPacks.
4. Run using the included launch configuration, see that exception isn't caught.
5. Install the [9-2020-q2 release from ARM](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads) and update the _ARM Toolchains Paths_ to point at it.
6. Clean and run, see that the exception is caught.

**Expected behaviour:** `Caught exception: My exception!` is printed to the console while running.

**Actual behaviour:** `std::terminate` handler is called.

### Versions

- xPack GNU Arm Embedded GCC v9.3.1-1.1
- Reproduced incorrect behavior on Ubuntu 18.04, Windows 10 2004, and macOS 10.15.6
- Reproduced correct behavior (with ARM distro) on Windows 10 2004 and macOS 10.15.6

Thank you for all your effort on this and adjacent projects -- I really like the xPack concept!

ilg-ul, 2020-08-24:
Thank you for your bug report. I'll investigate.

ilg-ul, 2020-08-25:
I confirm the issue.

Given the resulted size, it does not look like a newlib-nano simplification, it must be something else.

These days I have to finalise the migration of the project web to the Eclipse server, so it'll take a few more days to further investigate and fix it.

ilg-ul, 2020-08-26:
(This was a tricky one, and the challenge was too high to leave it for the next week...)

You were right to suspect a confusion with the `-nano` libraries. The library selection was correct, just that the non-nano libraries, more specifically libstdc++, were also built with support for exceptions disabled.

I fixed the scripts and ran a new build.

The new binaries are available for testing in the pre-release/test page:

https://github.com/xpack-dev-tools/pre-releases/releases/tag/test

(Arm binaries to come in a few more hours)

I checked the Mac binaries and they seem fine. Please check the binaries for your platforms of choice and let me know if they perform as expected, so I can proceed with the release.

willson556, 2020-08-26:
Tested on my Mac and it's working for me too! Thanks!

ilg-ul, 2020-08-26:
BTW, I also checked string exceptions, and they were caught:

```c++
  try {
  	  throw "exception";
  } catch (...) {
  	  trace_printf("Caught string exception\n");
  }
```

@fpistm, I guess your applications do not use exceptions, but if you can also test some cases on Windows/Linux, it would be very helpful.

fpistm, 2020-08-27:
Fine on windows.

ilg-ul, 2020-08-27:
Fixed in the 9.3.1-1.2 release.

closed 2020-08-27 by ilg-ul