xpack-dev-tools/arm-none-eabi-gcc-xpack #8: Build with python support

LnnrtS, 2020-10-05:
I am occasionally using [gdb dashboard](https://github.com/cyrus-and/gdb-dashboard) as a "gdb client". Unfortunately, the official toolchain is not built with python support, so I have to continue using the distro version (which is quite old).

I tried to build from source but without success. Do you think it would be feasible to enable python support in your build? Or does this only work in a distro context because it links against a certain python library that may not be present on the user's machine?

ilg-ul, 2020-10-05:
I'm not sure I understand your request.

There are both gdb-py and gdb-py3 binaries provided. They require Python 2.7 and Python 3.7 to run, as mentioned in the documentation.

ilg-ul, 2020-10-05:
> I am occasionally using gdb dashboard as a "gdb client"

I don't use it, but if you need a better way to interact with gdb, Eclipse has a very good integration, not to mention my plug-ins to see peripheral registers.

LnnrtS, 2020-10-05:
Thanks for the hint, I oversaw that file. Can you point me to the location in the documentation you are referring to? I cannot find it.

Using the correct binary brings me one step ahead but there is still something wrong. It cannot import the `math` module. `gdb-dashboard` imports a few modules, others (for example `ast`) work.

Minimal example:

```bash
$ ./arm-none-eabi-gdb-py3                                    
GNU gdb (xPack GNU Arm Embedded GCC, 64-bit) 8.3.1.20191211-git
...
(gdb) python
>import math
>end
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'math'
Error while executing Python code.
```

Running correct python version

```bash
$ python
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
```

ilg-ul, 2020-10-05:
- https://xpack.github.io/arm-none-eabi-gcc/python/

Frankly, when I first built gdb with Python support, I had a very poor understanding of Python internals, and everything looked like a huge mess, especially for Python 2.

Now, after building Meson with an embedded Python, I got a better view of it; it is still a mess, but in the future releases I'll try to make the best use of it.

The main problem is that Python can have its system libraries located in different places, and different distros do weird things. The Python executable obviously should be configured with these places in mind, but an application with python support (like gdb-py) does not know, and has to rely on some defaults, or on environment variables with the actual configuration.

Thus the recommendation to build a Python with the default settings, this should make life easier.

ilg-ul, 2020-10-05:
I took a second look at the specifics of gdb-py, and my previous guess proved correct, some distros set the PYTHONPATH in a specific way, which gdb-py does not know.

For example, on Ubuntu 18 (and I guess all Debian based distros), Python2 uses a sys.path like:

```python
['', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-x86_64-linux-gnu', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/usr/local/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages']
```

The unusual elements seem to be `plat-x86_64-linux-gnu` and `lib-tk`.

For Python3 apparently they simplified things and did not use those folders.

Anyway, in order to make gdb-py work you need to set:

- PYTHONPATH to a value based on `print(sys.path)`
- PTHONHOME to a value based on `print(sys.prefix)`

I'll update the documentation page to include these new findings.

LnnrtS, 2020-10-06:
Thanks for looking into this.

Unfortunately this does not fix the problem for me.
I guess the reason is `math` is a built-in module. But I cannot locate a dynamic library for it on the system nor see something getting loaded (via `strace`) that could fit. Will investigate a bit more..

ilg-ul, 2020-10-06:
I updated the previous code to set the environment.

The current strategy is to first try this.

If it fails, the final choice is to compile a 3.7 and install it as recommended in the [link](https://xpack.github.io/arm-none-eabi-gcc/python/), and then set the environment with the newly installed Python.

ilg-ul, 2020-10-06:
BTW, the math module normally is in `lib/python3.7/lib-dynload/math.cpython-3.7m-x86_64-linux-gnu.so`.

If not, it must be somewhere else in sys.path, thus setting the PYTHONPATH as recommended above might help.

LnnrtS, 2020-10-06:
I tried to set the environment variables also with the commands you suggest but this doesn't help.

Yes, I was looking at that place also but didn't find anything. [Debian packet search doesn't have it either](https://packages.debian.org/search?searchon=contents&keywords=math.cpython&mode=filename&suite=stable&arch=amd64) although the file could be named differently of course..

Maybe the module is statically linked to the `python` binary or something like that..?

ilg-ul, 2020-10-06:
> Yes, I was looking at that place also but didn't find anything. Debian packet search doesn't have it either although the file could be named differently of course.

The Debian approach of splitting distribution into multiple packages is weird (read stupid).

The file name must start with the module name, and, if a shared library, with some platform dependent suffix, it cannot have any name, since the python loader will not be able to identify it.

The packet search you referred to shows the module in the `python3-pygame` package.

> Maybe the module is statically linked to the python binary or something like that..?

I don't know, but I doubt the Python loader works with statically linked modules.

Try to install that package, but if it still fails with other missing modules, do yourself a favour and compile a Python 3.7 from sources, as recommended in the documentation (I use that page to copy/paste the few commands).

LnnrtS, 2020-10-06:
Installing `python3-pygame` didn't help.

I don't want to install a custom python version. Will stick with the current situation and keep eyes open for a cleaner longterm solution.

Thanks for your support so far!

ilg-ul, 2020-10-06:
> keep eyes open for a cleaner longterm solution.

Yeah, none as long as Debian's wise policy of splitting packages remains in place.

There is no clean way of using applications with embedded python support. Such applications have a hard dependency on a specific version of Python (3.7 in our case), and the chance of finding exactly that version on your system, and that version to be complete, is very low.

The alternates are to pack the Python standard library with the toolchain, as I did with Meson, or to create a separate binary xPack, with that Python version.

But given that compiling Python 3.7 is very easy, and the result is installed in `/usr/local/bin/python3.7`, thus does not interfere with the system python, for now I'll stick to recommending this solution.

ilg-ul, 2020-10-07:
> I don't want to install a custom python version.

Can you help me understand with your reasoning why compiling a Python from sources into `/usr/local` is not an acceptable solution?

I'm about to make new toolchain releases, and, if required, I can change the default location where gdb will search for the Python libraries. For a complete isolation, I'm considering using `/opt`, which is not in the PATH, so it should not interfere in any way with the other system installed tools.

But if you have a better solution, of course I can analyse it.

LnnrtS, 2020-10-07:
I have two reasons which are both not very convincing:
- I don't want to mess up my system as I don't have the time to properly understand all the steps involved
- It feels kind of wrong to replace working parts of my system just for adapting to the existing build process of these tools

I haven't really thought about this too much yet, but I see these options:
- Try to find the root of the problem and maybe fix it (couldn't make the native build work, yet)
- Check if [this ppa](https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa) works for my system

> For a complete isolation, I'm considering using /opt, which is not in the PATH, so it should not interfere in any way with the other system installed tools.
I think `/opt` is a good location to do that if you have root privileges

ilg-ul, 2020-10-07:
> I think `/opt` is a good location to do that if you have root privileges

Right. You served me my own argument ;-)
That's why all my tools use locations in the home folder. 

Then we'll use `$HOME/opt`; this does not require any changes in configuring the toolchain binaries, just a recommendation to use a script to set the Python environment before calling the debugger, something like:

```bash
mkdir -p "${HOME}/opt/bin"
cat <<'__EOF__' > "${HOME}/opt/bin/arm-none-eabi-gdb-py3.sh"
#!/usr/bin/env bash
PYTHONPATH="$($HOME/opt/bin/python3.7 -c 'import os; import sys; print(os.pathsep.join(sys.path))')" \
PYTHONHOME="$($HOME/opt/bin/python3.7 -c 'import sys; print(sys.prefix)')" \
arm-none-eabi-gdb-py3 "$@"
__EOF__
chmod +x "${HOME}/opt/bin/arm-none-eabi-gdb-py3.sh"
```

The script to build Python is:

```bash
python3_version="3.7.9"
mkdir -p "${HOME}/Downloads"
curl -L --fail -o "${HOME}/Downloads/Python-${python3_version}.tgz" https://www.python.org/ftp/python/${python3_version}/Python-${python3_version}.tgz 
rm -rf "${HOME}/Work/Python-${python3_version}"
mkdir -p "${HOME}/Work"
cd "${HOME}/Work"
tar xzf "${HOME}/Downloads/Python-${python3_version}.tgz" 
cd "${HOME}/Work/Python-${python3_version}"
bash ./configure --prefix="${HOME}/opt"
make 
make altinstall
```

On my VM it takes a few minutes to run, and it seems ok.

What do you think about this solution?

---

> I don't want to mess up my system as I don't have the time to properly understand all the steps involved

This is generally a good argument. In my case 'messing my system' means installing anything in the system path. I have lots of alternate environments, but all are in separate folders, and I have short shell aliases to activate them by adding their folders to the current terminal path.

> It feels kind of wrong to replace working parts of my system just for adapting to the existing build process of these tools

First, I doubt the build process of the toolchain can do anything to make the missing `math` Python module work as if it is present, except that provide an alternate Python standard library.

Then, if Python provides a module in their distribution, and the build procedure always builds it, but the distribution decides to not include it in the standard package, I think the problem is with the distribution, not with Python or with the toolchain.

> Try to find the root of the problem and maybe fix it 

That's clear, your distribution does not pack all Python modules in the standard package. What else do you suspect?

> (couldn't make the native build work, yet)

The build process is complex, it builds all libraries and even the toolchain requires several steps.

It is an interesting experience, but it is very time-consuming. Anyway, during development, do not build all multilibs, and do not run the tests. I'm not sure the native script has command line options for this, the distribution build has.

> Check if this ppa works for my system

Yes, some time ago I was also a bit more enthusiastic on using ppas and I considered them as alternates for a reproductive build environment, but later I realised they are a pain in the rear, and are better to be disregarded.

LnnrtS, 2020-10-15:
I can confirm that following these scripts fixes the `import math`
problem. Somehow for this particular problem using only one of the two
environment variables is sufficient.

> That's clear, your distribution does not pack all Python modules in the standard package. What else do you suspect?

Just to make sure I understand it correctly: Inside gdb there is a python interpreter independent from the system one but it tries to load packages from the system through the paths provided. So the open question to solve (maybe not inside this issue) is where debian is keeping the math module. Right?

ilg-ul, 2020-10-15:
> using only one of the two environment variables is sufficient.

Yes, if you build Python from sources the resulting folder structure matches the defaults used while building the Python run-time for GDB, and one definition might be enough, but in general, if you use the Python from the distribution, the Python modules may be split amongs multiple (weird) location, so you need the full PYTHONPATH, and sometimes, if you do not have the PYTHONHOME, you get an warning. So better use both of them.

> Inside gdb there is a python interpreter independent from the system one

That's correct. The Python interpreter is actually a shared library, like python37.so, so it is locked to a specific version, because different versions have different APIs. The first thought was to make GDB use the system library, which knows the location of the modules, so do not need to pass environment variables.

Unfortunately, Python distributions are not mandated to provide the shared library, the `python` program can be statically linked to the library, so GDB has no chance of finding it. The solution was to pack the shared library with GDB.

In addition to the shared library, a Python distribution includes two more things: a folder with the system `.py` modules and a folder (`lib-dynload`) with system native modules, as shared libraries. 

The system `.py` modules can be split amongst multiple folders, and the `PYTHONPATH` adjusted accordingly; this is what Ubuntu does, it has multiple locations, and you have to install multiple packages to get the full system library. Don't ask me why they're doing this, but it gave me quite a lot of headaches.

LnnrtS, 2020-10-15:
I see that there is are dynamic libraries for python3 installed on my system. When I replace the compiled library with a symlink to the system-one, everything works.

For me, this workaround is easier than compiling python. Do you think that would be easy to support? Trying to the load the system one and falling back to local one?

ilg-ul, 2020-10-15:
> Trying to the load the system one and falling back to local one?

How exactly would this mechanism work? 

When I build GDB, I set a DT_RPATH=$ORIGIN, which means the loader will start to search for libraries in the same folder as GDB. I don't know where your library is, to place its location in front of $ORIGIN, and even if you tell me the location, this setting will be used for all shared libraries required by GDB, and most probably it'll crash.

I think that changing the content of a binary xPack is not a very good workaround, especially since a future release of xpm will make it read only.

LnnrtS, 2020-10-15:
> When I build GDB, I set a DT_RPATH=$ORIGIN, which means the loader will start to search for libraries in the same folder as GDB. I don't know where your library is, to place its location in front of [...]

Actually it's even easier: It works as soon as I delete the local library. Seems to fall back to system one.

This makes me think its actually a bug because a local override of the library is provided but that override makes a few assumptions on the system that are neither checked nor enforced. I would say that either no local library should be provided or it should be done in way that ensures it will work.

> I think that changing the content of a binary xPack is not a very good workaround, especially since a future release of xpm will make it read only.

That was only meant as a test.

ilg-ul, 2020-10-15:
> Seems to fall back to system one.

That's easy to check, issue `ldd -v arm-none-eabi-gdb-py3` and the loader will tell you where it gets the libraries from.

> I would say that either no local library should be provided or it should be done in way that ensures it will work.

Feel free to suggest how to improve this. And be sure that the mechanism works on all supported platforms.

LnnrtS, 2020-10-15:
> Feel free to suggest how to improve this

- Make sure the local directory is searched *after* the system one. It looks like this is possible with `-Wl,--enable-new-dtags` or `DT_RUNPATH` (https://news.ycombinator.com/item?id=14222349)
- Ship full python with the toolchain and a wrapper as described in https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/issues/8#issuecomment-705057031
- Provide builds with and without the python library

> And be sure that the mechanism works on all supported platforms

I don't have the knowledge to judge that I am afraid

ilg-ul, 2020-10-15:
> Make sure the local directory is searched after the system one. It looks like this is possible with -Wl,--enable-new-dtags or DT_RUNPATH

Nope. For standalone package that's a wrong approach. The problem is that you want **only** the Python library to be searched in the system location. All other libraries should be searched in the current folder, otherwise the consistency of the build is compromised.

> Ship full python with the toolchain

This is possible, but will add some tens of MB to the binary, and it also has the disadvantage that if you want to install extra Python modules to be used with GDB, it'll be difficult, since the embedded Python will be read only, and you need to use an external folder, than you have to add it to PYTHONPATH each time you start GDB or the embedded Python.

> Provide builds with and without the python library

This simply doubles the maintenance effort. 

Overall it is much easier to install a custom Python in your home folder.

LnnrtS, 2020-10-15:
> if you want to install extra Python modules to be used with GDB, it'll be difficult, since the embedded Python will be read only, and you need to use an external folder, than you have to add it to PYTHONPATH each time you start GDB or the embedded Python.

That's why this would require the wrapper you posted earlier so it will at least use the system packages.

> but will add some tens of MB to the binary

That's the problem you get when you ship all the dependencies you need. It's basically the same as distribution package system vs snap. But if you start to go one way you have live with the downsides (in this case increase of size and bad system integration, just like snap). If you don't ship some dependencies you have to make sure they are met.

> This simply doubles the maintenance effort.

I might not have enough insight but deleting just a single file doesn't sound like a huge effort.

If nothing changes, the package is basically broken for debian which is the worst options of all in my opinion.

ilg-ul, 2020-10-15:
> If you don't ship some dependencies you have to make sure they are met.
> If nothing changes, the package is basically broken for debian which is the worst options of all in my opinion.

You are very right. 

I guess we'll have to pack the Python system library inside the toolchain package.
It is a bit of work, but I've already done it for meson, so it is feasible.

LnnrtS, 2020-10-15:
Ok, great. Let me know when I can help with testing it

ilg-ul, 2020-10-22:
I added a Python 3 instance and configured gdb-py3 to use it.

An experimental pre-release is available from

- https://github.com/xpack-dev-tools/pre-releases/releases/tag/experimental

To save some time, the binaries in this preliminary release do not include the multilibs, they'll be added in the final release.

It passed my initial tests, I'm waiting for Travis to run more comprehensive test, but it'll take some time, Travis is very slow at this time.

Please test gdb-py3 and let me know if it works for your use case.

ilg-ul, 2020-10-23:
Fixed in 9.3.1-1.4, on 2020-10-23.

closed 2020-10-23 by ilg-ul

ilg-ul, 2020-10-24:
https://xpack.github.io/blog/2020/10/23/arm-none-eabi-gcc-v9-3-1-1-4-released/

LnnrtS, 2020-10-24:
Yes, everything seems to be fine. Just didn't answer because the issue was already closed.

Thanks for fixing this! Having a portable cross gdb with working python will be useful for us.

ilg-ul, 2020-10-24:
Ok, great, thank you for your help.