xpack-dev-tools/arm-none-eabi-gcc-xpack #9: Native build fails: 'docker_images' not found

LnnrtS, 2020-10-07:
### Description

I followed the instructions for a native build but `build-native.sh` failed with

`./build-native.sh: Line 84: docker_images: Command not found.`

I expect that function should be defined in any of the common scripts but it is not

### Steps to Reproduce

```bash
git clone --recurse-submodules https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack.git
cd arm-none-eabi-gcc-xpack/scripts
./build-native.sh
```

I also installed docker and downloaded the image beforehand but as I understand I don't need them in a native build anyways..?

### Versions

`xpack` branch at 5d7df6285b38346cc03ec5da6b17ce0b25a72541
Debian 10.6

ilg-ul, 2020-10-07:
Oops! I haven't been using that script for long, and it seems it is not up-to-date.

ilg-ul, 2020-10-07:
I added those definitions, and the script starts on my mac, but I have no idea what are the prerequisites for it to run on any platform.

ilg-ul, 2020-10-07:
For the prerequisites, I suggest you take a look at the README-DEVELOP.md file in the openocd-xpack.git, they should be more or less the same.

LnnrtS, 2020-10-15:
That seems to have fixed it for me also. Thanks!

(The build still fails but that is issue is resolved)

ilg-ul, 2020-10-15:
I retain the idea that we need the prerequisites, probably in a central web page, and make all READMEs refer to it.

closed 2020-12-18 by ilg-ul