xpack-dev-tools/arm-none-eabi-gcc-xpack #10: Add meson cross file

LnnrtS, 2020-10-15:
[Meson cross files](https://mesonbuild.com/Cross-compilation.html) are used to specify a cross compilation tool chain or other cross compilation settings during configure-time. Multiple files can be layered

    meson setup build --cross-file /path/to/toolchain.conf --cross-file /path/to/mcu.conf

As described in the meson manual, cross files are searched for in fixed system locations so no absolute paths need to be specified. I think it would be useful for this xpack to install a cross file in such a location in order to enable a workflow similar to this:

    xpm install --global @xpack-dev-tools/arm-none-eabi-gcc@latest
    meson setup build --cross-file xpack-gcc-latest.conf --cross-file /my/custom/mcu.conf

On linux, `~/.local/share/meson/cross ` would be an appropriate location to store the cross file.

As an alternative, just having the cross-file inside the xpack would also be helpful

ilg-ul, 2020-10-15:
Yes, I was thinking about this, since I also need toolchain descriptions for the xPack Builder, and I know CMake too has something similar.

We'll think what would be a good location for them. 

For now I suggest you copy them to your project, to be safe, and we'll simplify things later.

LnnrtS, 2020-10-15:
Just for the record: '~' is not expanded for the paths given in cross files so absolute paths will be needed

ilg-ul, 2020-10-15:
> ~ is not expanded for the paths given in cross files

What exactly does this mean? 

Isn't `~` expanded by the shell before passing the options to meson?

ilg-ul, 2020-10-15:
BTW, one possible solution that I considered so far is to add a separate package for each toolchain, for example `arm-none-eabi-gcc-toolchain-defs` where we put together various definitions for the supported tools (meson, CMake, xpbuild, etc).

The definitions for xpbuild can be quite elaborated, since, in theory at least, it is possible to define how a GUI (like Eclipse or Visual Studio Code) would display the settings page, and this requires a bit more details than the basic definitions used to generate the ninja/make files.

LnnrtS, 2020-10-15:
> Isn't ~ expanded by the shell before passing the options to meson?

Meson parses the cross-file and uses the provided paths. There is no shell involved.

> BTW, one possible solution that I considered so far is to add a separate package for each toolchain, for example arm-none-eabi-gcc-toolchain-defs where we put together various definitions for the supported tools (meson, CMake, xpbuild, etc).

I would say its cleaner to keep the descriptions and the content that they describe in a single place

ilg-ul, 2020-10-15:
> Meson parses the cross-file and uses the provided paths. There is no shell involved.

Then I misunderstood the issue. You probably mean paths _inside_ the cross files. 

> its cleaner to keep the descriptions and the content that they describe in a single place

In general, yes, but what if you want to use a custom toolchain, that you compiled yourself, or got from another place? 

Since such a definitions for `arm-none-eabi-gcc` should be usable for all `arm-none-eabi-gcc`, not only for the xPack version.

I don't know if you noticed, but I'm doing my best to avoid making xPacks mandatory, and always give the users alternate solutions, even if they require less automation, such that those who are allergic to using xpm to install xPacks will not reject everything a priory.

That's why you can install binary xPacks by manually downloading archives, replace source xPacks with git submodules, or why xpbuild will be able to build any projects, not only those made of xPacks. Just that in each case it is a bit more work to do manually.

LnnrtS, 2020-10-15:
> Then I misunderstood the issue. You probably mean paths inside the cross files.

yes

> but what if you want to use a custom toolchain, that you compiled yourself, or got from another place?

How do you want to support that case? How do you know the paths that should be written inside the cross file? The only option I see is to provide the cross file for everything that is accessible via `PATH`. I don't have an opinion yet on where such a file should be stored.

ilg-ul, 2020-10-15:
Yes, although I do not recommend it for production projects, most of the users still reach the toolchain via the system PATH.

Does meson expand environment variables inside the cross files?

LnnrtS, 2020-10-15:
I see it like this: the toolchain should describe itself if possible so it can be used with build systems. Just like a library ships a `CMakeLists.txt. This obviously only works when you have a fixed installation location (so you can write that into the cross file). For xpacks this is the case so creating a cross file should be done during package installation. I don't know if that's doable but this is how I would expect it to work.

The case for toolchains in the `PATH` is a different question I would say.

> Does meson expand environment variables inside the cross files?

I think environment variables are never used throughout meson.

ret394, 2020-11-24:
The opposite concept for enhancement is a more apt label. To avoid misleading other and yourself. :-)

ilg-ul, 2020-11-24:
That's too subtle for me :-(

ret394, 2020-11-24:
Then add me as co-maintainer and I'll explain it.

you're really welcome
ret394

ilg-ul, 2020-11-24:
@ret394, I'll think about your offer to become a co-maintainer and educator.

Till then, you're blocked.

closed 2020-12-18 by ilg-ul