xpack-dev-tools/arm-none-eabi-gcc-xpack #11: Add cmake toolchain file

LnnrtS, 2020-11-16:
I propose to add this minimal cmake toolchain file in the root directory of the built binaries

```cmake
set(TOOLCHAIN_BASE_DIR ${CMAKE_CURRENT_LIST_DIR}/bin)

set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_PROCESSOR arm)
set(CMAKE_TRY_COMPILE_TARGET_TYPE "STATIC_LIBRARY")

set(CMAKE_C_COMPILER   "${TOOLCHAIN_BASE_DIR}/arm-none-eabi-gcc")
set(CMAKE_CXX_COMPILER "${TOOLCHAIN_BASE_DIR}/arm-none-eabi-g++")
find_program(CMAKE_SIZE NAMES arm-none-eabi-size    HINTS ${TOOLCHAIN_BASE_DIR})
find_program(CMAKE_GDB  NAMES arm-none-eabi-gdb-py3 HINTS ${TOOLCHAIN_BASE_DIR})
```
`size` and `gdb` need to be added manually because cmake doesn't track these tools automatically

ilg-ul, 2020-11-16:
Could you provide more details? What should be the file name and where exactly should it be located?

And could you also document the use case? 

At first sight I'm not convinced that the right place for such a file is the GCC distribution; since the original GCC build procedures do not create such a file, and most distributions, including the major one, from Arm, do not add it separately, I would say that CMake projects have to handle such details privately.

If I add it, you configure your projects to use, and, at a certain point you'll want to do a build with the distribution compiler, what will happen?

LnnrtS, 2020-11-16:
As this file uses `${CMAKE_CURRENT_LIST_DIR}` it should be located "next to the readme".

When you configure a cross cmake project for the first time you need to specify such a toolchain file. It can be located anywhere with any name I think (though `.cmake` is used usually). The file is parsed as a regular cmake script and its purpose is to setup the toolchain and (in our case) enable a cross build. After the first configuration run, the file is not read anymore but the toolchain properties are stored inside the configured project (as cache variables).

If you want to change the toolchain, you have to re-configure the whole project from scratch. I guess there is a way to store the current configuration and re-apply it to the project (once reconfigured) but from cmake's point if view it needs to start from a clean slate.

I am also surprised why those files are not shipped anywhere but I cannot find any downside for that, because
- its a standard format
- this file doesn't make any assumptions on anything outside this package

I guess system packages expect to get found via `PATH` so you could get away with a generic toolchain file using `PATH` inside the cmake project. For all other toolchains, having the toolchain file inside the project would mean making assumptions about the installation location of the toolchain which is unfortunate.

closed 2020-12-18 by ilg-ul