xpack-dev-tools/arm-none-eabi-gcc-xpack #12: new m1 Mac is not supported

vandarri, 2020-12-30:
### Disclaimer

The xPack GNU Arm Embedded GCC is only a binary distribution of
the [GNU Arm Embedded Toolchain](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm),
and does not intend to add new functionality, or to fix existing problems.

If you have questions/suggestions related to the procedure used to
build/publish the binary distribution, continue reading the next sections.

If you have problems related to Arm GCC functionality, please use the
official [Launchpad tracker](https://bugs.launchpad.net/gcc-arm-embedded/).

### Prerequisites

Before entering a new ticket, please consider the following:

- [ ] read the documentation pages carefully (`share/doc/pdf/gcc/gcc.pdf`)
- [ ] check the [xPack Forums](https://www.tapatalk.com/groups/xpack/) for similar problems
- [ ] check the [GitHub Issues](https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/issues/)

If you still could not find an answer, if you have interesting use
cases, if you have custom configurations, and generally if you have
any experience that you would like to share with others, use the
[xPack Forums](https://www.tapatalk.com/groups/xpack/), the
GNU Arm Embedded GCC section.

If you are convinced that you identified a bug related to the binary
distribution, not to GNU Arm Embedded GCC itself (if you have doubts,
use the forum), or you have a pertinent suggestion how to enhance the
xPack GNU Arm Embedded GCC distribution, continue and register a new issue.

### Description

Hi I'm using new m1 MacBook Air with Homebrew
It seems there is no darwin-arm64 version on release page.
Is there any plan or beta version for apple silicon?

### Steps to Reproduce

I used command like below.
xpm install --global @xpack-dev-tools/arm-none-eabi-gcc@latest

**Expected behaviour:** install xpack

**Actual behaviour:** error message occurred
error: Platform darwin-arm64 not supported.

### Versions

Big Sur

Please understand that without being able to reproduce the bug we cannot
identify your problem.

---

Note: Remove unnecessary text after reading, before entering your ticket.

ilg-ul, 2020-12-30:
Support for the new Macs is planned, but currently I do not have access to such a development machine.

vandarri, 2020-12-30:
Thank you for your work! I hope it will be soon.

ilg-ul, 2020-12-30:
You're welcome.

I plan to buy a Mac Mini M1, but it is not yet available here, so all I can do is wait for Apple to update their offer.

If you know someone with a Mac Mini or a MacBook Pro that can allow me remote ssh access, I can try to prepare the build environment and run the build scripts, until a permanent solution will be available.

yennster, 2021-03-18:
@ilg-ul Found this thread through googling issues with arm-none-eabi-g++ compiling on an Apple M1 computer, I am actually using https://www.scaleway.com/en/ Scaleway to remote log into an M1 machine that I pay for by the hour, just FYI

ilg-ul, 2021-03-18:
Thank you, Jenny, for the suggestion. I took a quick look at the link, but I'm not sure I understand the details. 

Due to the specifics of the xPack builds, and the lack of a mechanism similar to Docker, I currently need a persistent virtual machine, I cannot use a fresh instance for each build. Is this supported?

yennster, 2021-03-18:
@ilg-ul Hi yes, I spent a few hours using a persistent virtual machine on Scaleway today, it gives you the option to connect via SSH or VNC remote desktop, the remote desktop has a full Mac M1 GUI desktop environment for your use

yennster, 2021-03-18:
(By the way, this isn't a Scaleway ad or anything, I just thought it was useful as I am also needing to debug M1 compile issues haha)

ilg-ul, 2021-03-19:
> I spent a few hours using a persistent virtual machine on Scaleway today

I created an account to experiment with Scaleway, and I noticed that the price is $0.1/hour, with a min 24 hours.

What I did not understand is how much it costs simply to store a dormant virtual machine image. My typical use case is very sporadic, from several minutes from now and then, lets say every month for short binary xPacks, to several hours for toolchains for 4-5 times per year. But preparing the xPack Build Box environment takes many hours, and it is not reasonable to do it before every build.

Crazor, 2021-11-16:
Since this issue was closed a long time ago, I'd like to ask: Did I miss the actual solution? Is this still in the "planned" stage or is there support for arm64 now?

ilg-ul, 2021-11-16:
I got a Mac Mini M1 from MacStadium a few days ago, and I'm working on porting the build scripts to it right now.

There are some problems though, GCC is not yet ported properly to Apple Silicon, there is only a fork of GCC 11.1, which I already built, and there are plans to upstream the changes for GCC 12.0 and probably 11.3, but this will take probably some months to be released.

I'll try to do my best of the existing tools, but I don't know yet how difficult it'll be.

Crazor, 2021-11-16:
I'll gladly help out with testing, I have a Mac Mini and a MacBook Air with Apple Silicon on hand. 
For the time being, I tried to run the Intel version of the toolchain via Rosetta, however I failed to install via XPM, because XPM complains about missing support for arm64, even though I ran it through the `arch` wrapper

ilg-ul, 2021-11-16:
I have no experience with the `arch` wrapper, but, knowing xpm, I don't think it can be tricked. xpm itself is portable, and works well with Apple Silicon, but it has its own logic with architectures, if it does not find the correct entries in `package.json`, it fails.

But if you install the Intel archive in a folder and handle paths manually, does the compiler run under Rosetta?

ilg-ul, 2021-11-16:
The first attempt to build the toolchain failed in GCC with _Undefined symbols for architecture arm64:_ which most probably is caused by GCC 10 not being ready for Apple Silicon.

The first GCC which adds support for Apple Silicon is 11.1. I'll try to take a look at the patches, perhaps some can be back-ported to 10, but I cannot guarantee anything for now. :-(

ilg-ul, 2021-11-17:
After getting [help](https://github.com/iains/gcc-darwin-arm64/issues/49), I managed to build an experimental macOS arm64 toolchain:

- <https://github.com/xpack-dev-tools/pre-releases/releases/tag/experimental>

Please try it, since currently I don't have the development environment to do it myself.

ilg-ul, 2021-11-19:
I just installed the latest Eclipse 2021-12 on a Mini M1, and I could test the toolchain, it seems fine.

@choeoliv, @Crazor, @yennster, could you confirm that the new toolchain works in your environment too?

After manually unpacking the archive, you have to remove the quarantine atribute, otherwise you might not be able to run the binaries (`xattr -dr com.apple.quarantine xpack-*`).

Crazor, 2021-11-22:
That build seems to be working just fine, successfully compiled a bunch of projects and ran them on different boards.

ilg-ul, 2021-11-22:
Thank you! I'm updating all packages; so far only QEMU is problematic.

If everything is fine, I'll release them, probably in about 1 week.

kammce, 2021-12-07:
Were these released?

ilg-ul, 2021-12-07:
If everything goes according to plan, tomorrow should be public.

kammce, 2021-12-07:
Oh whoa! Awesome to hear!

ilg-ul, 2021-12-07:
> I failed to install via XPM, because XPM complains about missing support for arm64, even though I ran it through the arch wrapper

For the records, to get the architecture, xpm asks nodeJs, which uses the definition hardcoded at build time, thus the `arch` command does not help in this case.

ilg-ul, 2021-12-07:
Fixed on 2021-12-07.

<https://xpack.github.io/blog/2021/12/07/arm-none-eabi-gcc-v10-3-1-2-3-released/>

closed 2021-12-07 by ilg-ul

hawkinchina, 2024-11-05:
Can support Macos 10.13.4 ?    the official arm-none-eabi-gcc need 10.15 at least of Macos.

ilg-ul, 2024-11-05:
The x64 binaries are targeted for macOS 10.13.