xpack-dev-tools/arm-none-eabi-gcc-xpack #16: Compile error: `cannot execute 'cc1': execvp: No such file or directory`

robamu, 2021-12-08:
Hello,

I am trying to compile an assembler file into an archive but I have an issue doing this with the xPacks compiler:

```sh
[ARM none EABI] rmueller@power-pinguin:~/Rust/cortex-m-rt(master)$ ./assemble.sh 
+ crate=cortex-m-rt
+ rm -f 'bin/*.a'
+ arm-none-eabi-gcc -g -c -march=armv6s-m asm.S -o bin/cortex-m-rt.o
arm-none-eabi-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
compilation terminated.
```

Any ideas how to solve this?

The error can be reproduced by cloning https://github.com/rust-embedded/cortex-m-rt and then running the
command `arm-none-eabi-gcc -g -c -march=armv6s-m asm.S -o bin/cortex-m-rt.o` (and ensuring that the binary folder is of course added in the system path)

ilg-ul, 2021-12-08:
What version? What platform?

robamu, 2021-12-08:
Sorry, here are some additional information:

OS / Platform: Ubuntu 21.04
Version: arm-none-eabi-gcc (xPack GNU Arm Embedded GCC x86_64) 10.3.1 20210824 (release)

ilg-ul, 2021-12-08:
That version string only shows the Arm 20210824 release, please provide the full xPack package version.

robamu, 2021-12-08:
Provided that as well.

robamu, 2021-12-08:
Hmm, that was what I was hoping to see as well. Maybe there is something wrong with my PATH setup. I'll report back soon

robamu, 2021-12-08:
Missclick

ilg-ul, 2021-12-08:
There is a good reason why I invented xpm and an automatic way of managing paths.

If you add an xPack action to the `package.json` (like `"xyz" : "bash assemble.sh"`), you can run it directly with `xpm run xyz`, and do not have to deal with paths at all.

ilg-ul, 2021-12-08:
One suggestion: could you run the failing compile command with `-v`, to see exactly what is happening?

robamu, 2021-12-09:
Okay, the issue was that I used a bash script like this to enter a temporary environment

```sh
#!/bin/bash -i
echo "Entering ARM (NONE EABI) environment"
export PATH="~/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin":$PATH
export CONSOLE_PREFIX="[ARM (NONE EABI)]"
exec /bin/bash
```
the purpose is that you can use exit to remove the path. But that process seems to be problematic for this task. If I load the path manually like you did, it works.

This is also something I commonly need for other purposes, but it always worked there.
For example, we use a custom CMake build system where we set up the build system manually once and build the software with `cmake --build . -j` once this is done. We still require full control of the build system, also to stay independent of a specific IDE. To do all of this, I still need to load the environment once with something like `export PATH="~/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin":$PATH` before setting up a build system for cross-compiling . CMake caches the path after that and having the path in envrionemnt is only necessary when setting up the build system, but it would be very nice if one could load the binary path of a package temporarily conveniently. Is there some way to do that?

ilg-ul, 2021-12-09:
> export PATH="~/.local..."

I tried using `~` in this context, and in my environment (macOS) the substitution did not happen. I suggest you use $HOME explicitly, this works always.

> CMake ...

I had no problems to use CMake and meson. The recommended way is to use xpm to install the toolchain locally to your project, as seen before, then to start the build as I did, or even better via xpm run, and things are as simple as they can be.

if you already installed the toolchain with xpm install -g, installing locally to a project means only creating some links, it is very light. And the advantage is that you can use different toolchain versions in different projects. Not to mention that you don't have to worry about paths at all.

> It would be very nice if one could load the binary path of a xpm package like arm-none-eabi temporarily conveniently. Is there some way to do that?

I'm afraid I do not understand this question. :-(

robamu, 2021-12-09:
It really was the tilde, it works with `$HOME` now. Thanks for that hint :)

My question was whether there is some way to convenienty invoke this command

```sh
export PATH="$HOME/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin":$PATH
```

to set up the path via `xpm` somehow. I'll try your way of installing it locally to the project as well.

ilg-ul, 2021-12-09:
> It really was the tilde, it works with $HOME now. Thanks for that hint :)

You're welcome, but if the xPack toolchain was not really in your path, what arm-none-eabi-gcc did it run (and fail)?

> whether there is some way to convenienty invoke this command ... to set up the path via xpm somehow.

Yes, this is one of the things `xpm run` does for you automatically. 

Actually this is done in two steps, `xpm install` without `--global` adds links from the `<project>/xpacks/.bin` to the actual `$HOME/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/...`, executables, then `xpm run` always prepends `<project>/xpacks/.bin` to the path and runs the desired command in this environment.

Take a look at an example generated by the project template:

- <https://xpack.github.io/vscode/quick-start/>

You don't have to use VS Code, you can also run the commands by hand, if you desire so.

The template project is:

- <https://github.com/xpack/hello-world-template-xpack>

I'll have similar templates for embedded projects soon, but for a functional example you can use the native Hello World as well.

ilg-ul, 2021-12-09:
I tried this for your project, and here is the result.

I first added the `xpack.actions` map to `package.json`:

```json
{
  ...
  "xpack": {
    "minimumXpmRequired": "0.10.8",
    "actions": {
      "assemble": "bash assemble.sh"
    }
  }
}
```

A run look like this:

```console
ilg@ubu20:~/cortex-m-rt$ xpm run assemble
> bash assemble.sh
+ crate=cortex-m-rt
+ rm -f bin/thumbv6m-none-eabi.a bin/thumbv7em-none-eabi.a bin/thumbv7em-none-eabihf.a bin/thumbv7m-none-eabi.a bin/thumbv8m.base-none-eabi.a bin/thumbv8m.main-none-eabi.a bin/thumbv8m.main-none-eabihf.a
+ arm-none-eabi-gcc -g -c -march=armv6s-m asm.S -o bin/cortex-m-rt.o
+ ar crs bin/thumbv6m-none-eabi.a bin/cortex-m-rt.o
+ arm-none-eabi-gcc -g -c -march=armv7-m asm.S -o bin/cortex-m-rt.o
+ ar crs bin/thumbv7m-none-eabi.a bin/cortex-m-rt.o
+ arm-none-eabi-gcc -g -c -march=armv7e-m asm.S -o bin/cortex-m-rt.o
+ ar crs bin/thumbv7em-none-eabi.a bin/cortex-m-rt.o
+ arm-none-eabi-gcc -g -c -march=armv7e-m asm.S -DHAS_FPU -o bin/cortex-m-rt.o
+ ar crs bin/thumbv7em-none-eabihf.a bin/cortex-m-rt.o
+ arm-none-eabi-gcc -g -c -march=armv8-m.base asm.S -o bin/cortex-m-rt.o
+ ar crs bin/thumbv8m.base-none-eabi.a bin/cortex-m-rt.o
+ arm-none-eabi-gcc -g -c -march=armv8-m.main asm.S -o bin/cortex-m-rt.o
+ ar crs bin/thumbv8m.main-none-eabi.a bin/cortex-m-rt.o
+ arm-none-eabi-gcc -g -c -march=armv8-m.main -DHAS_FPU asm.S -o bin/cortex-m-rt.o
+ ar crs bin/thumbv8m.main-none-eabihf.a bin/cortex-m-rt.o
+ rm bin/cortex-m-rt.o
ilg@ubu20:~/cortex-m-rt$ 
```

You can see the adjusted path if you invoke xpm with more verbosity:

```console
ilg@ubu20:~/cortex-m-rt$ xpm run assemble -v
xPack manager - run package/configuration specific action

CWD=/home/ilg/cortex-m-rt
PATH=/home/ilg/cortex-m-rt/xpacks/.bin:/home/ilg/.nvm/versions/node/v16.13.1/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

Invoking 'bash assemble.sh'...
+ crate=cortex-m-rt
...
```

robamu, 2021-12-09:
`You're welcome, but if the xPack toolchain was not really in your path, what arm-none-eabi-gcc did it run (and fail)?`

Actually it was in the path and I was able to use the binaries. But some internal escaping issue (at least thats what I'm guessing) made lead to this error where `cc1` was not found.

Thanks a lot for the hints, they help a lot. This will probably make the installation and setup instructions for our CMake projects a bit easier. You can close this if you want, all my questions have been answered :)

ilg-ul, 2021-12-09:
> Actually it was in the path and I was able to use the binaries. But some internal escaping issue (at least thats what I'm guessing) made lead to this error where cc1 was not found.

This is what worries me, and that's why I would like to better understand what happened, since such cases should not be possible.

I would appreciate if you could recreate the environment where it failed, and run the same command with `-v`, to see some internals.

>  installation and setup instructions for our CMake projects a bit easier

BTW, I also have binary xPacks with CMake and ninja, adding them as project dependencies not only make installation and setup easier (all you have to do is issue `xpm install` in the project folder and all dependencies are satisfied automatically), but also guarantees that the project uses exactly the desired versions, and multiple projects may use different versions. These reproducible builds are really important when running tests.

robamu, 2021-12-10:
Okay, I loaded the path with `~` and this the the output of `arm-none-eabi-gcc -v -g -c -march=armv6s-m asm.S -o bin/test.o`

```txt
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
Target: arm-none-eabi
Configured with: /Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/sources/gcc/configure 
--prefix=/Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/linux-x64/install/arm-none-eabi-gcc 
--infodir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/linux-x64/install/arm-none-eabi-gcc/share/doc/info
 --mandir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/linux-x64/install/arm-none-eabi-gcc/share/doc/man --htmldir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/linux-x64/install/arm-none-eabi-gcc/share/doc/html 
--pdfdir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/linux-x64/install/arm-none-eabi-gcc/share/doc/pdf 
--build=x86_64-unknown-linux-gnu --host=x86_64-unknown-linux-gnu --target=arm-none-eabi --with-pkgversion='xPack GNU Arm Embedded GCC\x2C 64-bit' --enable-languages=c,c++ --disable-mingw-wildcard --enable-plugins 
--enable-lto --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath 
--disable-libssp --disable-libstdcxx-pch --disable-nls --disable-threads --disable-tls --with-gnu-as --with-gnu-ld 
--with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/linux-x64/install/arm-none-eabi-gcc/arm-none-eabi 
--with-native-system-header-dir=/include --with-multilib-list=rmprofile --disable-build-format-warnings --with-system-zlib
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.1 20201103 (release) (xPack GNU Arm Embedded GCC, 64-bit) 
COLLECT_GCC_OPTIONS='-v' '-g' '-c'  '-o' 'bin/test.o' '-mfloat-abi=soft' '-mthumb' '-march=armv6s-m'
 cc1 -E -lang-asm -quiet -v -imultilib thumb/v6-m/nofp -iprefix ../lib/gcc/arm-none-eabi/10.2.1/ -D__USES_INITFINI__ asm.S -mfloat-abi=soft -mthumb -march=armv6s-m -g -fworking-directory -fno-directives-only -o /tmp/ccti6kxw.s
arm-none-eabi-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
compilation terminated.
```

And this is the output using `$HOME`:

```txt
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
Target: arm-none-eabi
Configured with: /Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/sources/gcc/configure --prefix=/Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/linux-x64/install/arm-none-eabi-gcc --infodir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/linux-x64/install/arm-none-eabi-gcc/share/doc/info --mandir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/linux-x64/install/arm-none-eabi-gcc/share/doc/man --htmldir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/linux-x64/install/arm-none-eabi-gcc/share/doc/html --pdfdir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/linux-x64/install/arm-none-eabi-gcc/share/doc/pdf --build=x86_64-unknown-linux-gnu --host=x86_64-unknown-linux-gnu --target=arm-none-eabi --with-pkgversion='xPack GNU Arm Embedded GCC\x2C 64-bit' --enable-languages=c,c++ --disable-mingw-wildcard --enable-plugins --enable-lto --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/Host/home/ilg/Work/arm-none-eabi-gcc-10.2.1-1.1/linux-x64/install/arm-none-eabi-gcc/arm-none-eabi --with-native-system-header-dir=/include --with-multilib-list=rmprofile --disable-build-format-warnings --with-system-zlib
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.2.1 20201103 (release) (xPack GNU Arm Embedded GCC, 64-bit) 
COLLECT_GCC_OPTIONS='-v' '-g' '-c'  '-o' 'bin/test.o' '-mfloat-abi=soft' '-mthumb' '-march=armv6s-m'
 /home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../libexec/gcc/arm-none-eabi/10.2.1/cc1 -E -lang-asm -quiet -v -imultilib thumb/v6-m/nofp -iprefix /home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../lib/gcc/arm-none-eabi/10.2.1/ -isysroot /home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../arm-none-eabi -D__USES_INITFINI__ asm.S -mfloat-abi=soft -mthumb -march=armv6s-m -g -fworking-directory -fno-directives-only -o /tmp/ccpqkeOy.s
ignoring duplicate directory "/home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../lib/gcc/../../lib/gcc/arm-none-eabi/10.2.1/include"
ignoring nonexistent directory "/home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../arm-none-eabi/usr/local/include"
ignoring duplicate directory "/home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../lib/gcc/../../lib/gcc/arm-none-eabi/10.2.1/include-fixed"
ignoring duplicate directory "/home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../lib/gcc/../../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include"
ignoring duplicate directory "/home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../arm-none-eabi/include"
#include "..." search starts here:
#include <...> search starts here:
[... cut ...] ib/gcc/arm-none-eabi/10.2.1/:/home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../lib/gcc/:/home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/lib/:/home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../arm-none-eabi/lib/
COLLECT_GCC_OPTIONS='-v' '-g' '-c'  '-o' 'bin/test.o' '-mfloat-abi=soft' '-mthumb' '-march=armv6s-m'
```

ilg-ul, 2021-12-10:
Thank you for the test. The difference is in the line invoking cc1:

```console
 cc1 -E -lang-asm -quiet -v -imultilib thumb/v6-m/nofp -iprefix ../lib/gcc/arm-none-eabi/10.2.1/ -D__USES_INITFINI__ asm.S -mfloat-abi=soft -mthumb -march=armv6s-m -g -fworking-directory -fno-directives-only -o /tmp/ccti6kxw.s
arm-none-eabi-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
```

versus

```console
 /home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../libexec/gcc/arm-none-eabi/10.2.1/cc1 -E -lang-asm -quiet -v -imultilib thumb/v6-m/nofp -iprefix /home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../lib/gcc/arm-none-eabi/10.2.1/ -isysroot /home/rmueller/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.2.1-1.1.2/.content/bin/../arm-none-eabi -D__USES_INITFINI__ asm.S -mfloat-abi=soft -mthumb -march=armv6s-m -g -fworking-directory -fno-directives-only -o /tmp/ccpqkeOy.s
```

In other words, with the correct path, `arm-none-eabi-gcc` is able to correctly identify the actual location of `cc1`, and invokes it with the full path.

In the first case this does not happen, and `gcc` defaults to invoking `cc1` by name and also does not pass many options.

The logic for identifying `cc1` is quite complicated, and there are many ways it can fail, for example in the Windows binaries the implementation of the system function `access()` behaves differently from Unix, and a mingw-w64 hack is required.

On Linux there should be other details that do not match...

On the other hand, the output you presented shows you are using a very old 10.2.1-1.1, although you mentioned using 10.3.1-2.3.1.

This makes me think that you actually **do** have the old one somewhere in the system path, which is totally not recommended.

Could you add a `which arm-none-eabi-gcc` to your script and run it in the case with the tilde path? Also an `echo $PATH` would help.