xpack-dev-tools/arm-none-eabi-gcc-xpack #17: v10.3.1-2.3 deleted?

fpistm, 2022-03-24:
Hi @ilg-ul 
It seems the GH release v10.3.1-2.3 has been removed? Why?

ilg-ul, 2022-03-24:
oops!

it was definitely a mistake, I'll fix it as soon as possible.

ilg-ul, 2022-03-24:
It looks like I inadvertently pressed the wrong button and I re-published that release, which changed it to draft. :-(

I'm re-running the tests to be sure, but I guess it is ok now.

fpistm, 2022-03-24:
Thanks for the quick answer and fix 👏 
Hope the sha256 is the same else the current release of the STM32 core will be deprecated.
I'm waiting user feedback about this.

ilg-ul, 2022-03-24:
Normally the sha256 should be the same, my understanding is that only the draft flag was set, the rest of the content was preserved, anyway I did not have to actually re-load the files, they were there, but not visible.

fpistm, 2022-03-24:
Ah OK. So it should be fine.
I close this issue. Thank you again 👍

closed 2022-03-24 by fpistm

ilg-ul, 2022-03-24:
You're welcome!