xpack-dev-tools/arm-none-eabi-gcc-xpack #18: Not compatible with Raspberry Pi 0/1 (Armv6)

jzobac, 2022-03-31:
Hi,
I have getting errors when trying to run gcc

10.2.1 - segmentation fault
10.3.1 - illegal instruction

Probably it must be compiled with flags:
--with-arch=armv6 --with-float=hard --with-fpu=vfp

With best regards
JZ

ilg-ul, 2022-03-31:
Since you do not provide details on how you invoked the compiler and the source code, I am not abe to reproduce yoru errors.

Could you try to compile exactly the same source files, with exactly the same command lines, but with the GNU Arm Embedded Toolchain distribution?

prso, 2022-04-23:
I have the same problem from a Raspberry Pi (1) Model B with 512 MB of RAM and Raspberry Pi OS last version (bullseye) upgraded today.

Steps to reproduce it:

`$ wget  https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/download/v10.2.1-1.1/xpack-arm-none-eabi-gcc-10.2.1-1.1-linux-arm.tar.gz`
`(...)`

`$ tar -xzf xpack-arm-none-eabi-gcc-10.2.1-1.1-linux-arm.tar.gz`

`$ xpack-arm-none-eabi-gcc-10.2.1-1.1/bin/arm-none-eabi-gcc -v`
`Segmentation fault`

ilg-ul, 2022-04-23:
What does `uname -a` report for your system?

ilg-ul, 2022-04-23:
I checked and indeed the first generation of RPi used ARMv6.

The current xPack binaries were created on ARMv7, and this explains the segmentation fault.

I'll check if the build environment can be recreated on an ARMv6 machine, but I'm not sure that this can be achieved with reasonable efforts.

ilg-ul, 2022-04-24:
Unfortunately my first attempt to reconfigure the build environment to use an old Raspbian 9 failed.

There might be alternate solutions, but I don't have now the resources to further investigate, so I'm afraid that the xPack binaries will remain supported from ARMv7 up.

prso, 2022-04-25:
I'm a bit confused, why do you need Raspbian 9?

The Raspberry Pi 1 ARMv6 (first model) is perfectly compatible with the last Raspbian 11 (bullseye) (32-bit version) (officially known as Raspberry Pi OS by the Raspberry Pi Foundation).

 I'm running it without problems.

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 armv6l GNU/Linux
```
```
pi@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
```
pi@raspberrypi:~ $ file /bin/ls
/bin/ls: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, BuildID[sha1]=81004d065160807541b79235b23eea0e00a2d44e, for GNU/Linux 3.2.0, stripped
```

ilg-ul, 2022-04-25:
I know this is arbitrarily, but this is it for now: the minimum requirement for all GNU/Linux xPack binaries is Ubuntu 18, which uses GLIBC 2.27.

Raspberry Pi OS (actually Debian) buster uses GLIBC 2.28, and binaries built on it are not guaranteed to run on Ubuntu 18.

Thus the need to go one step further back in time, which is Raspbian 9.

ilg-ul, 2022-04-25:
BTW, are you sure that with 512 MB of RAM you can really use the system for building real-life project? I mean more complicated that a blinky. From my experience, a minimum would by 1 GB, which means RPi 2, thus my minimum platform set to ARMv7. However I might be wrong, since I had no access to an early RPi.

prso, 2022-04-25:
I honestly don't know because the compiler is not workig in my RPi 1. ;)

I have already used another RPi (model 3) for compiling [game-and-wtach-retro-go](https://github.com/kbeckmann/game-and-watch-retro-go) that in fact has 1 GB. I have seen people in the forums that use a RPi zero for that, but mybe they compile in a different machine and use the RPi zero only for flashing, I don't know.

Thanks anyway @ilg-ul

jzobac, 2022-04-25:
I have upgraded to Debian 11 and no problem to compile g&w firmware for Zelda edition and 4MB memory it's about 3h at @950MHz. Debian 11 has all needed functions, so maybe this is just waste of time. But will be good to say officialy, that this is not suitable for RPI 1.

ilg-ul, 2022-04-25:
>  But will be good to say officially, that this is not suitable for RPI 1

Right, I updated the documentation.

closed 2022-05-30 by ilg-ul