xpack-dev-tools/arm-none-eabi-gcc-xpack #22: update Arm GCC 11.3.Rel1

dsm, 2022-08-19:
Arm GNU Toolchain
Version 11.3.Rel1

Released: August 8, 2022

ilg-ul, 2022-08-19:
Thank you, I'll make a new release.

Do you know of any method to get notifications when Arm makes new releases?

dsm, 2022-08-19:
Thanks, I don't know, I check manually but maybe using a CI script to parse the web site every day(maybe week) for new release

ilg-ul, 2022-08-19:
Yes, this might be a solution. Improve it to compare the string with a known one and fail the CI job if different, to trigger an email. And add the GitHub Action definition.

ilg-ul, 2022-08-28:
- <https://xpack.github.io/blog/2022/08/28/arm-none-eabi-gcc-v11-3-1-1-1-released/>

Please notify me on future Arm releases, since I don't monitor the Arm site.

closed 2022-08-28 by ilg-ul