xpack-dev-tools/arm-none-eabi-gcc-xpack #25: Newlib version header file

ilg-ul, 2022-10-19:
### Discussed in https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/discussions/24

<div type='discussions-op-text'>

<sup>Originally posted by **jasr93** October 19, 2022</sup>
Hi everyone,

I upgraded my toolchain from 11.2.1-1.2.2 to 11.3.1-1.1.2 and I have a build error related to Newlib version.

The _newlib_version.h header file provided by the toolchain has the following content:

/* _newlib_version.h.  Generated from _newlib_version.hin by configure.  */
/* Version macros for internal and downstream use. */
#ifndef _NEWLIB_VERSION_H__
#define _NEWLIB_VERSION_H__ 1

/* The newlib version in string format. */
#define _NEWLIB_VERSION "@NEWLIB_VERSION@"

/* The newlib major version number. */
#define __NEWLIB__ @NEWLIB_MAJOR_VERSION@

/* The newlib minor version number. */
#define __NEWLIB_MINOR__ @NEWLIB_MINOR_VERSION@

/* The newlib patch level. */
#define __NEWLIB_PATCHLEVEL__ @NEWLIB_PATCHLEVEL_VERSION@

#endif /* !_NEWLIB_VERSION_H__ */

However, it seems that the placeholders have not been changed to their corresponding numbers and gcc complains with "error: token "@" is not valid in preprocessor expressions"

I was able to verify that in the 11.2.1-1.2.2 toolchain, this file is formatted correctly with numbers instead of placeholders.

Thanks in advance

</div>

ilg-ul, 2022-10-19:
I confirm that both the arm and aarch64 toolchains have this problem.

@jasr93, could you check if the Arm original toolchains are also affected?

jasr93, 2022-10-19:
Yes,
I have downloaded arm-gnu-toolchain-11.3.rel1 from the ARM Developer website and the header files have the same problem.

ilg-ul, 2022-10-19:
hmmm... in this case it'll be more complicated, since the xPack distribution follows the official one, and the problem should be first fixed upstream.

could you report the bug upstream and cc to me, so I can follow the issue?

jasr93, 2022-10-19:
OK,

I will create a case in ARM GNU forum

nicreuss, 2022-12-17:
Bug tracker link with updated status: https://bugs.linaro.org/show_bug.cgi?id=5894

ilg-ul, 2022-12-17:
Good, so in one of the future releases it'll be fixed.

closed 2023-02-05 by ilg-ul