xpack-dev-tools/arm-none-eabi-gcc-xpack #29: where should I add -g option for Newlib

hellowoad, 2023-07-11:
hi, I'm new to arm toolchain. I want build the toolchain with a debugable Newlib and  Newlib-nano. is it possible? if it's possible, where can I add -g option?

hellowoad, 2023-07-11:
sorry, my question may be stupid and obvious, 😭

ilg-ul, 2023-07-11:
Don't worry, the question is not stupid, occasionally I need the debuggable libraries too, and I planned to address this in the next release.

The `build.sh` script already accepts a `--debug` option, but it probably does not apply to newlib too.

If you experiment with the scripts, please note that it's not only that the `-g` must be added, but also the `strip` must be disabled, otherwise the debug info is removed at the end.

I'll leave this ticket open, and address it in the near future (but no clear date set).

ilg-ul, 2023-07-16:
I took a look at the scripts, and the official way to add debug info to the build is to use `--develop --debug` when invoking the build.

I did such a run, but, for unknown reasons, the debugger could still not step through the library functions, so probably there are some other details that I missed and need to be fixed.

I'll try to address this issue during the next few days, since I also have to fix a problem with timezone definitions.

hellowoad, 2023-07-17:
Thank you for your work！Expect this feature to be implemented

ilg-ul, 2023-07-17:
It should be functional now.

Support for debug builds was added via the `build-develop-debug` or `docker-build-develop-debug` xPack actions, or by directly invoking the build script with `--develop --debug`.

I checked the macOS build and I was able to step by step inside newlib. I expect the Linux build to behave similarly. The Windows build (which runs on Linux) is a bit more complex and takes more time, I did not have time to check it.

Please let me know if it works for you too.

hellowoad, 2023-07-18:
wow, great job! I will try to build and debug it on my apple silicon MacBook tonight.

ilg-ul, 2023-07-18:
Great! Thank you for confirming.

closed 2023-07-18 by ilg-ul