xpack-dev-tools/arm-none-eabi-gcc-xpack #32: Turn off --enable-newlib-mb option

ladislas, 2023-08-21:
Hi there! 

Starting with GCC 11, the ARM toolchain team introduced the new option `--enable-newlib-mb` to provide a unified release for all A,R,M-class processors.

This change had a dramatic impact on flash/ram consumption for small embedded projects like ours.

Detailed discussion on the impact of the change and how/why is was introduced can be found here: https://bugs.linaro.org/show_bug.cgi?id=5929 (bug open by me)

The option is there in the manifest:

https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/blob/87b0d36d8a0c410a87b2cb4f336abdf73c981c88/extras/12.3.rel1/12.3.rel1-darwin-x86_64-arm-none-eabi-manifest.txt#L11C217-L11C236

I'm wondering if it would be possible for xpack to provide a release without the `--enable-newlib-mb` option?

Thanks a lot for your help :)

ilg-ul, 2023-08-21:
This is a hard one.

Unless there is something terribly wrong, this distribution tries to remain fully compatible with the Arm official distribution. As such, it also use `--enable-newlib-mb` for the full libraries, and does not use it for the nano libraries.

If Arm acknowledges that there is something wrong with this option, and removes it, we'll catch it in the next release.

ladislas, 2023-08-22:
I understand, even though **+26'238 bytes** of for flash consumption is a lot when just upgrading toolchain.

Would you welcome a PR to add an "experimental" build for the toolchain without the option while also keeping the official one?

ilg-ul, 2023-08-22:
> +26'238 bytes for flash consumption is a lot

well, yes, it is not a joy, but I don't know if it is justified or not.

> a PR to add an "experimental" build

if you are interested in maintaining extra builds, you can fork the project and run your own builds; I don't have resources to do this.

I advise you to either use the nano version, or to further investigate where the extra bytes come from, and, if you are sure that code is never used, to define empty functions with the same name.

closed 2023-09-21 by ilg-ul