xpack-dev-tools/arm-none-eabi-gcc-xpack #35: Position-independent code conflicts with libgcc.a

anjiahao1, 2023-12-28:
### Disclaimer

The xPack GNU Arm Embedded GCC is only a binary distribution of
the [Arm GNU Toolchain](https://developer.arm.com/Tools%20and%20Software/GNU%20Toolchain),
and does not intend to add new functionality, or to fix existing problems.

If you have questions/suggestions related to the procedure used to
build/publish the binary distribution, continue reading the next sections.

If you have problems related to Arm GCC functionality, please use the
official [Launchpad tracker](https://bugs.launchpad.net/gcc-arm-embedded/).

### Prerequisites

Before entering a new ticket, please consider the following:

- [ ] read the documentation pages carefully (`share/doc/pdf/gcc/gcc.pdf`)
- [ ] check the [xPack Forums](https://www.tapatalk.com/groups/xpack/) for similar problems
- [ ] check the [GitHub Issues](https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/issues/)

If you still could not find an answer, if you have interesting use
cases, if you have custom configurations, and generally if you have
any experience that you would like to share with others, use the
[xPack Forums](https://www.tapatalk.com/groups/xpack/), the
GNU Arm Embedded GCC section.

If you are convinced that you identified a bug related to the binary
distribution, not to GNU Arm Embedded GCC itself (if you have doubts,
use the forum), or you have a pertinent suggestion how to enhance the
xPack GNU Arm Embedded GCC distribution, continue and register a new issue.

### Description

[Description of the bug or feature]

### Steps to Reproduce

I'm trying to us`e -fpic -mno-pic-data-is-text-relative -mpic-register=r9 `to compile address-independent code and run it on a CM0 device, and everything works fine. But until the float division was run, an error occurred. After my debug, I found that the division implemented by libgcc.a in the compiler did not use` -fpic -mno-pic-data-is-text-relative -mpic-register=r9` , causing the device to crash. Any ideas? Looking forward to your reply

**Expected behaviour:** [What you expected to happen]
libgcc.a can also be position independent
**Actual behaviour:** [What actually happened]
libgcc.a address is related and is not affected by compilation options

Or 
1. How to recompile libgcc.a with compile options ?
2. compile libgcc.a in source code form instead of .a during compilation?

### Versions

- [Arm GCC version]
13.1
- [operating system]
ubuntu23.04

Please understand that without being able to reproduce the bug we cannot
identify your problem.

---

Note: Remove unnecessary text after reading, before entering your ticket.

ilg-ul, 2023-12-28:
Did you read the message that you posted above?

anjiahao1, 2023-12-28:
> Did you read the message that you posted above?

Yes, I searched a lot of information, but I couldn't find any relevant solutions.

ilg-ul, 2023-12-28:
Does the original Arm GNU Toolchain 13.1 distribution include an address-independent libgcc.a?

anjiahao1, 2023-12-28:
> Does the original Arm GNU Toolchain 13.1 distribution include an address-independent libgcc.a?

No, I checked that versions 11, 12, and 13 all have only one copy of libgcc.a. If I can recompile a copy of libgcc.a myself, this problem can theoretically be solved.

ilg-ul, 2023-12-28:
Sorry, but what makes you think that you are reporting an issue related to this project?

anjiahao1, 2023-12-28:
`arm-none-eabi-gcc-xpack` should theoretically also encounter this problem. I couldn't find how to recompile libgcc.a on the Internet. This .a seems to be related to the architecture.

ilg-ul, 2023-12-28:
If you confirm that this project behaves the same as the original Arm GCC, and you still think that this is not correct, you must report the issue to Arm.

closed 2023-12-28 by ilg-ul