xpack-dev-tools/arm-none-eabi-gcc-xpack #41: how to modify compile parameter？

snikeguo, 2025-09-17:
1.My embedded project requires TLS support enabled (--enable-tls). However, most pre-built arm-none-eabi-gcc toolchains are configured with --disable-tls. I attempted to rebuild the toolchain from source (version: xpack-arm-none-eabi-gcc-13.3.1-1.1) by modifying all build scripts in the build-assets/extras directory, replacing --disable-tls with --enable-tls. After compilation, the generated toolchain still reports --disable-tls in its configuration details:
```
Using built-in specs.
COLLECT_GCC=D:\Soft\ArmGcc\xpack-arm-none-eabi-gcc-13.3.1-1.1\bin\arm-none-eabi-gcc.exe
COLLECT_LTO_WRAPPER=D:/Soft/ArmGcc/xpack-arm-none-eabi-gcc-13.3.1-1.1/bin/../libexec/gcc/arm-none-eabi/13.3.1/lto-wrapper.exe
Target: arm-none-eabi
Configured with: /home/david/projects/xbb_work/arm-none-eabi-gcc-xpack-13.3.1-1.1/build-assets/build/win32-x64/sources/gcc/configure --prefix=/home/david/projects/xbb_work/arm-none-eabi-gcc-xpack-13.3.1-1.1/build-assets/build/win32-x64/application --with-sysroot=/home/david/projects/xbb_work/arm-none-eabi-gcc-xpack-13.3.1-1.1/build-assets/build/win32-x64/application/arm-none-eabi --with-native-system-header-dir=/include --infodir=/home/david/projects/xbb_work/arm-none-eabi-gcc-xpack-13.3.1-1.1/build-assets/build/win32-x64/x86_64-w64-mingw32/install/share/info --mandir=/home/david/projects/xbb_work/arm-none-eabi-gcc-xpack-13.3.1-1.1/build-assets/build/win32-x64/x86_64-w64-mingw32/install/share/man --htmldir=/home/david/projects/xbb_work/arm-none-eabi-gcc-xpack-13.3.1-1.1/build-assets/build/win32-x64/x86_64-w64-mingw32/install/share/html --pdfdir=/home/david/projects/xbb_work/arm-none-eabi-gcc-xpack-13.3.1-1.1/build-assets/build/win32-x64/x86_64-w64-mingw32/install/share/pdf --build=x86_64-pc-linux-gnu --host=x86_64-w64-mingw32 --target=arm-none-eabi --disable-nls --disable-shared --disable-threads --disable-tls --enable-checking=release --enable-languages=c,c++ --enable-mingw-wildcard --with-gmp=/home/david/projects/xbb_work/arm-none-eabi-gcc-xpack-13.3.1-1.1/build-assets/build/win32-x64/x86_64-w64-mingw32/install --with-newlib --with-pkgversion='xPack GNU Arm Embedded GCC x86_64' --with-gnu-as --with-gnu-ld --with-system-zlib --with-headers=yes --enable-multilib --with-multilib-list=aprofile,rmprofile
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 13.3.1 20240614 (xPack GNU Arm Embedded GCC x86_64)
```
Expected Behavior:​​
The toolchain should be compiled with --enable-tls to support TLS functionality.
Steps to Reproduce:​​

Download xpack-arm-none-eabi-gcc-13.3.1-1.1 source.
Modify all instances of --disable-tls to --enable-tls in build-assets/extras scripts.
Compile the toolchain following the build instructions:
```
xpm run install
xpm run docker-prepare --config win32-x64
xpm run docker-build --config win32-x64
```
Check configuration with arm-none-eabi-gcc -v.

2.I would like to customize the GCC version string from:
xPack GNU Arm Embedded GCC x86_64
to:
xPack GNU Arm Embedded GCC x86_64, built by XXX
Could you please advise on how to properly modify the build scripts to achieve both TLS enablement and version string customization?

snikeguo, 2025-09-17:
> > Most microcontrollers nowadays run an RTOS
> 
> Which one do you know to support the ARM ABI TLS model? (like using r9, /, , etc)`.tdata``.tbss``__tls_get_addr()`

I speculate that the toolchain might generate assembly code like:
ldr r1, [r9, #offset]  // Get TLS value,r9 is tp reg
where the offset is determined during the linking stage.

ilg-ul, 2025-09-17:
> Currently, NuttX handles the .tbss and .tdata sections during thread creation

Ok, thank you for the info, I'll take a look at how NuttX does this, and experiment with it on my µOS++, which already supports standard C++ threads, but without TLS.

snikeguo, 2025-09-17:
[image]

[image]

[image]
My steps:
```
cd xbb-xpack-helper // into helper dir
xpm install 
cd arm-none-eabi-gcc-xpack-13.3.1-1.1 //into gcc dir
xpm link @xpack-dev-tools/xbb-helper-xpack
```
Modified configuration(build-assets/package.json):
"@xpack-dev-tools/xbb-helper": "^3.0.13"
But the generated config.log still shows --disable-tls

snikeguo, 2025-09-17:
> https://xpack-dev-tools.github.io/arm-none-eabi-gcc-xpack/docs/developer/

I just downloaded xbb helper 3.0.13. Now I have deleted 3.0.13 and plan to use the default version 3.0.10. I have modified the configuration with --enable-tls, recompiled it, and I'll give it a try.
Thanks.

ilg-ul, 2025-09-17:
Again, I have no idea what you are doing. The scripts allow to restart a broken build, and if you want to restart with a clean slate you have to run the deep-clean action. See the Maintainer page for details.

snikeguo, 2025-09-17:
> Not that it matters, but where in the documentation did you read about running these commands with sudo? On the contrary, the documentation explicitly asks to **not** install npm/xpm with root rights.
> 
> The whole build runs very well with regular user rights.

The issue has been resolved. I no longer have problems when compiling without Docker... The generated parameters are: --disable-threads --enable-tls.

snikeguo, 2025-09-17:
> Ok, assuming the toolchain is ok now, can you build your app with the standard C11 or C++11 ? If so, what do you have to do in your RTOS to allocate the TLS segment for each thread?`_Thread_local``thread_local`

The compilation is not yet complete; it looks like it will take 3 to 4 hours.
I'm using the pthread library and not using C++. Once the compilation is finished, I can write a demo and post it here.

ilg-ul, 2025-09-18:
Most probably the linker picked up libraries from the system folders, instead of those compiled from sources.

This might happen when you do not follow the build instructions.

snikeguo, 2025-09-18:
```
readelf -d /home/david/projects/xbb_work/arm-none-eabi-gcc-xpack-13.3.1-1.1/build-assets/build/linux-x64/application/libexec/libstdc++.so.6.0.32 

Dynamic section at offset 0x263d80 contains 30 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [ld-linux-x86-64.so.2]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x000000000000000e (SONAME)             Library soname: [libstdc++.so.6]
 0x000000000000000c (INIT)               0x9c000
 0x000000000000000d (FINI)               0x1ccac4
 0x0000000000000019 (INIT_ARRAY)         0x25b550
 0x000000000000001b (INIT_ARRAYSZ)       136 (bytes)
 0x000000000000001a (FINI_ARRAY)         0x25b5d8
 0x000000000000001c (FINI_ARRAYSZ)       8 (bytes)
 0x000000006ffffef5 (GNU_HASH)           0x328
 0x0000000000000005 (STRTAB)             0x2ded8
 0x0000000000000006 (SYMTAB)             0x9230
 0x000000000000000a (STRSZ)              308127 (bytes)
 0x000000000000000b (SYMENT)             24 (bytes)
 0x0000000000000003 (PLTGOT)             0x266000
 0x0000000000000002 (PLTRELSZ)           25272 (bytes)
 0x0000000000000014 (PLTREL)             RELA
 0x0000000000000017 (JMPREL)             0x95990
 0x0000000000000007 (RELA)               0x7cc40
 0x0000000000000008 (RELASZ)             101712 (bytes)
 0x0000000000000009 (RELAENT)            24 (bytes)
 0x000000006ffffffc (VERDEF)             0x7c388
 0x000000006ffffffd (VERDEFNUM)          51
 0x000000006ffffffe (VERNEED)            0x7ca90
 0x000000006fffffff (VERNEEDNUM)         4
 0x000000006ffffff0 (VERSYM)             0x79278
 0x000000006ffffff9 (RELACOUNT)          949
 0x0000000000000000 (NULL)               0x0

```

no rpath.
I am currently using xbb 3.1.13, but the original version was 3.0.10. I will revert to version 3.0.10 and modify the configuration from --disable-tls to --enable-tls, then recompile it.

ilg-ul, 2025-09-18:
FYI: All binaries (executables & libraries) compiled from sources include RPATH, since they must be adjusted in a post-processing step to ensure the toolchain is standalone and runs from any location.

snikeguo, 2025-09-18:
> Great!
> 
> Did you understand how this works? Who allocates the TLS sections, `pthread_create()`?
> 
> What platform are you running the program on? Do you have something special in the linker script for this to work?

Let's discuss the issue you mentioned later. I believe there might be a problem with the GCC toolchain's output. According to the AAPCS specification, R3 should be pushed or preserved. However, the generated assembly instructions did not preserve R3. I traced inside the __aeabi_read_tp function and confirmed that it modifies the R3 register. Therefore, I think this might be a bug in GCC?
```
// static
inline Thread * ThreadStore::RawGetCurrentThread()
{
    Thread* th= (Thread *) &tls_CurrentThread;
     C0017D7A   LDR            R3, =0x001A5528               ; [PC, #24] [0xC0017D94] 
     C0017D7C   ADD            R3, PC
     C0017D7E   LDR            R3, [R3]      ; R3 should be pushed or preserved   
     C0017D80   BL             __aeabi_read_tp               ; 0xC009F40C
     C0017D84   MOV            R2, R0
     C0017D86   ADD            R3, R2
     C0017D88   STR            R3, [R7, #4]
    return th;
}
```
C0017D7E   LDR            R3, [R3]      ; R3 should be pushed or preserved

ilg-ul, 2025-09-18:
> R3 should be pushed or preserved

I confess that I do not fully understand this code, so I cannot comment on this.

I suggest you compile the same code on an aarch64 linux and compare the results.

On the other hand, the arm-none-eabi-gcc code is maintained by Arm, and if they never enabled TLS, it is possible that the code was never fully tested.

snikeguo, 2025-09-18:
C0017D7A   LDR            R3, =0x001A5528               ; [PC, #24] [0xC0017D94] 
     C0017D7C   ADD            R3, PC
     C0017D7E   LDR            R3, [R3]      ; R3 should be pushed or preserved   
now R3 is 0x20 ,R3 is offset value

C0017D80   BL             __aeabi_read_tp               ; 0xC009F40C
get tdata base of current thread
 C0017D84   MOV            R2, R0
     C0017D86   ADD            R3, R2
R3=R3+R2, //If R3 has not been tampered with, then the value in R3 should be the address of the TLS variable.

 STR            R3, [R7, #4]
th :address of the TLS variable.

snikeguo, 2025-09-18:
Where should I submit a GCC issue?

ilg-ul, 2025-09-18:
In the GCC Bugzilla, see the GCC documentation.

snikeguo, 2025-11-06:
I need to recompile the libraries included with the GCC toolchain (for example libgcc.a, libm.a, libsupc++.a) to add the -fno-short-enums compiler option. Which file should I modify to change the compile options for these libraries?  Is it binutils-cross.sh?